balena-os/balena-beaglebone #25: mmc errors

floion, 2016-06-30:
Observed while deploying an app on the beaglebone black:

root@beaglebone:~# [  533.797044] mmcqd/1: page allocation failure: order:2, mode:0x2204020
[  533.803549] CPU: 0 PID: 94 Comm: mmcqd/1 Not tainted 4.4.9+ #1
[  533.809407] Hardware name: Generic AM33XX (Flattened Device Tree)
[  533.815568] [<c001afa0>] (unwind_backtrace) from [<c0015308>] (show_stack+0x20/0x24)
[  533.823357] [<c0015308>] (show_stack) from [<c054bd64>] (dump_stack+0x94/0xa8)
[  533.830629] [<c054bd64>] (dump_stack) from [<c015e204>] (warn_alloc_failed+0xec/0x128)
[  533.838588] [<c015e204>] (warn_alloc_failed) from [<c0161804>] (__alloc_pages_nodemask+0x900/0xac0)
[  533.847688] [<c0161804>] (__alloc_pages_nodemask) from [<c01a2998>] (cache_alloc_refill+0x340/0x558)
[  533.856867] [<c01a2998>] (cache_alloc_refill) from [<c01a31b0>] (__kmalloc+0x16c/0x198)
[  533.864910] [<c01a31b0>] (__kmalloc) from [<c05dc230>] (edma_prep_slave_sg+0xa8/0x340)
[  533.872875] [<c05dc230>] (edma_prep_slave_sg) from [<c0811208>] (omap_hsmmc_request+0x434/0x52c)
[  533.881712] [<c0811208>] (omap_hsmmc_request) from [<c07f7fb4>] (__mmc_start_request+0x50/0xe0)
[  533.890455] [<c07f7fb4>] (__mmc_start_request) from [<c07f81a4>] (mmc_start_request+0x160/0x298)
[  533.899282] [<c07f81a4>] (mmc_start_request) from [<c07f97ec>] (mmc_start_req+0x29c/0x3ec)
[  533.907586] [<c07f97ec>] (mmc_start_req) from [<c08094e0>] (mmc_blk_issue_rw_rq+0xc0/0xb1c)
[  533.915976] [<c08094e0>] (mmc_blk_issue_rw_rq) from [<c080a1a4>] (mmc_blk_issue_rq+0x268/0x4f4)
[  533.924714] [<c080a1a4>] (mmc_blk_issue_rq) from [<c080bcac>] (mmc_queue_thread+0xd0/0x19c)
[  533.933110] [<c080bcac>] (mmc_queue_thread) from [<c00635bc>] (kthread+0xec/0x104)
[  533.940724] [<c00635bc>] (kthread) from [<c0010a40>] (ret_from_fork+0x14/0x34)
[  533.947975] Mem-Info:
[  533.950268] active_anon:44582 inactive_anon:2101 isolated_anon:0
[  533.950268]  active_file:32754 inactive_file:33078 isolated_file:0
[  533.950268]  unevictable:0 dirty:2406 writeback:4010 unstable:0
[  533.950268]  slab_reclaimable:7715 slab_unreclaimable:1871
[  533.950268]  mapped:6584 shmem:2140 pagetables:334 bounce:0
[  533.950268]  free:883 free_pcp:47 free_cma:0
[  533.983803] Normal free:3532kB min:2824kB low:3528kB high:4236kB active_anon:178328kB inactive_anon:8404kB active_file:131016kB inactive_file:132312kB unevictable:0kB isolated(anon):0kB isolated(file):0kB pro
[  534.029776] lowmem_reserve[]: 0 0 0
[  534.033303] Normal: 273*4kB (UE) 305*8kB (U) 0*16kB 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB 0*8192kB = 3532kB
[  534.045543] 67977 total pagecache pages
[  534.049398] 0 pages in swap cache
[  534.052725] Swap cache stats: add 0, delete 0, find 0/0
[  534.057968] Free swap  = 0kB
[  534.060856] Total swap = 0kB
[  534.063745] 130560 pages RAM
[  534.066633] 0 pages HighMem/MovableOnly
[  534.070480] 5628 pages reserved
[  534.073630] 6144 pages cma reserved
[  534.077345] edma 49000000.edma: edma_prep_slave_sg: Failed to allocate a descriptor
[  534.085083] omap_hsmmc 481d8000.mmc: prep_slave_sg() failed
[  534.090701] omap_hsmmc 481d8000.mmc: MMC start dma failure
[  534.565156] mmcblk0: unknown error -1 sending read/write command, card status 0x900
[  534.573018] blk_update_request: I/O error, dev mmcblk0, sector 2946304
[  534.579617] blk_update_request: I/O error, dev mmcblk0, sector 2946312
[  534.586263] blk_update_request: I/O error, dev mmcblk0, sector 2946320
[  534.592861] blk_update_request: I/O error, dev mmcblk0, sector 2946328
[  534.599439] blk_update_request: I/O error, dev mmcblk0, sector 2946336
[  534.606035] blk_update_request: I/O error, dev mmcblk0, sector 2946344
[  534.612629] blk_update_request: I/O error, dev mmcblk0, sector 2946352
[  534.619206] blk_update_request: I/O error, dev mmcblk0, sector 2946360
[  534.625797] blk_update_request: I/O error, dev mmcblk0, sector 2946368
[  534.632389] blk_update_request: I/O error, dev mmcblk0, sector 2946376
[  534.641574] BTRFS error (device mmcblk0p6): bdev /dev/mmcblk0p6 errs: wr 1, rd 0, flush 0, corrupt 0, gen 0
[  534.652128] BTRFS error (device mmcblk0p6): bdev /dev/mmcblk0p6 errs: wr 2, rd 0, flush 0, corrupt 0, gen 0
[  534.662821] mmcqd/1: page allocation failure: order:2, mode:0x2204020
[  534.669302] CPU: 0 PID: 94 Comm: mmcqd/1 Not tainted 4.4.9+ #1
[  534.675157] Hardware name: Generic AM33XX (Flattened Device Tree)
[  534.681315] [<c001afa0>] (unwind_backtrace) from [<c0015308>] (show_stack+0x20/0x24)
[  534.689103] [<c0015308>] (show_stack) from [<c054bd64>] (dump_stack+0x94/0xa8)
[  534.696376] [<c054bd64>] (dump_stack) from [<c015e204>] (warn_alloc_failed+0xec/0x128)
[  534.704336] [<c015e204>] (warn_alloc_failed) from [<c0161804>] (__alloc_pages_nodemask+0x900/0xac0)
[  534.713436] [<c0161804>] (__alloc_pages_nodemask) from [<c01a2998>] (cache_alloc_refill+0x340/0x558)
[  534.722614] [<c01a2998>] (cache_alloc_refill) from [<c01a31b0>] (__kmalloc+0x16c/0x198)
[  534.730658] [<c01a31b0>] (__kmalloc) from [<c05dc230>] (edma_prep_slave_sg+0xa8/0x340)
[  534.738621] [<c05dc230>] (edma_prep_slave_sg) from [<c0811208>] (omap_hsmmc_request+0x434/0x52c)
[  534.747459] [<c0811208>] (omap_hsmmc_request) from [<c07f7fb4>] (__mmc_start_request+0x50/0xe0)
[  534.756201] [<c07f7fb4>] (__mmc_start_request) from [<c07f81a4>] (mmc_start_request+0x160/0x298)
[  534.765028] [<c07f81a4>] (mmc_start_request) from [<c07f97ec>] (mmc_start_req+0x29c/0x3ec)
[... 53 lines cut ...]
[  535.848809] [<c080bcac>] (mmc_queue_thread) from [<c00635bc>] (kthread+0xec/0x104)
[  535.856422] [<c00635bc>] (kthread) from [<c0010a40>] (ret_from_fork+0x14/0x34)
[  535.863674] Mem-Info:
[  535.865968] active_anon:44587 inactive_anon:2101 isolated_anon:0
[  535.865968]  active_file:32735 inactive_file:33101 isolated_file:0
[  535.865968]  unevictable:0 dirty:1863 writeback:4159 unstable:0
[  535.865968]  slab_reclaimable:7714 slab_unreclaimable:1874
[  535.865968]  mapped:6598 shmem:2140 pagetables:335 bounce:0
[  535.865968]  free:883 free_pcp:38 free_cma:0
[  535.899502] Normal free:3532kB min:2824kB low:3528kB high:4236kB active_anon:178348kB inactive_anon:8404kB active_file:130940kB inactive_file:132404kB unevictable:0kB isolated(anon):0kB isolated(file):0kB pro
[  535.945476] lowmem_reserve[]: 0 0 0
[  535.949002] Normal: 281*4kB (UE) 301*8kB (U) 0*16kB 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB 0*8192kB = 3532kB
[  535.961239] 67984 total pagecache pages
[  535.965095] 0 pages in swap cache
[  535.968421] Swap cache stats: add 0, delete 0, find 0/0
[  535.973664] Free swap  = 0kB
[  535.976552] Total swap = 0kB
[  535.979441] 130560 pages RAM
[  535.982329] 0 pages HighMem/MovableOnly
[  535.986176] 5628 pages reserved
[  535.989326] 6144 pages cma reserved
[  535.993019] edma 49000000.edma: edma_prep_slave_sg: Failed to allocate a descriptor
[  536.000720] omap_hsmmc 481d8000.mmc: prep_slave_sg() failed
[  536.006350] omap_hsmmc 481d8000.mmc: MMC start dma failure
[  536.400034] mmcblk0: unknown error -1 sending read/write command, card status 0x900

agherzan, 2016-08-23:
@floion Could this be the same issue we experience on other boards?

floion, 2016-09-01:
Can you link me to the other issue you are referring to @agherzan ?

floion, 2016-09-01:
Also, today when trying to deploy an app, I got this:

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.9+ (builder@b5d3d6d5eaa8) (gcc version 5.2.0 (GCC) ) #1 SMP Wed Aug 31 15:10:07 UTC 2016
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: TI AM335x BeagleBone Black
[    0.000000] cma: Reserved 24 MiB at 0x9e000000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 130560
[    0.000000] free_area_init_node: node 0, pgdat c1065c80, node_mem_map df96d000
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 130560 pages, LIFO batch:31
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon )
[    0.000000] PERCPU: Embedded 13 pages/cpu @df924000 s24192 r8192 d20864 u53248
[    0.000000] pcpu-alloc: s24192 r8192 d20864 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 129408
[    0.000000] Kernel command line: console=ttyO0,115200n8 root=PARTUUID=c63c0048-02 rw rootfstype=ext4 rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 474416K/522240K available (11265K kernel code, 960K rwdata, 3924K rodata, 736K init, 909K bss, 23248K reserved, 24576K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
                   lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0xc0008000 - 0xc0edd77c   (15190 kB)
                     .init : 0xc0ede000 - 0xc0f96000   ( 736 kB)
                     .data : 0xc0f96000 - 0xc1086158   ( 961 kB)
                      .bss : 0xc1089000 - 0xc116c7d0   ( 910 kB)
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000]  RCU restricting CPUs from NR_CPUS=2 to nr_cpu_ids=1.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=32, nr_cpu_ids=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] IRQ: Found an INTC at 0xfa200000 (revision 5.0) with 128 interrupts
[    0.000000] OMAP clockevent source: timer2 at 24000000 Hz
[    0.000015] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
[    0.000037] clocksource: timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000048] OMAP clocksource: timer1 at 24000000 Hz
[    0.000375] Console: colour dummy device 80x30
[    0.000402] WARNING: Your 'console=ttyO0' has been replaced by 'ttyS0'
[    0.000410] This ensures that you still see kernel messages. Please
[    0.000416] update your kernel commandline.
[    0.000437] Calibrating delay loop... 996.14 BogoMIPS (lpj=4980736)
[    0.088912] pid_max: default: 32768 minimum: 301
[    0.089036] Security Framework initialized
[    0.089049] Yama: becoming mindful.
[    0.089090] AppArmor: AppArmor disabled by boot time parameter
[    0.089262] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.089274] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.090114] Initializing cgroup subsys io
[    0.090146] Initializing cgroup subsys memory
[    0.090186] Initializing cgroup subsys devices
[    0.090204] Initializing cgroup subsys freezer
[    0.090227] Initializing cgroup subsys net_cls
[    0.090242] Initializing cgroup subsys perf_event
[    0.090256] Initializing cgroup subsys net_prio
[    0.090309] CPU: Testing write buffer coherency: ok
[    0.090369] ftrace: allocating 36144 entries in 106 pages
[    0.192703] CPU0: thread -1, cpu 0, socket -1, mpidr 0
[... 498 lines cut ...]
[ 3177.712009] BTRFS warning (device mmcblk0p6): Skipping commit of aborted transaction.
[ 3177.719873] ------------[ cut here ]------------
[ 3177.724554] WARNING: CPU: 0 PID: 551 at /yocto/resin-board/build/tmp/work-shared/beaglebone/kernel-source/fs/btrfs/transaction.c:1746 btrfs_commit_transaction+0xa44/0xd60()
[ 3177.740026] BTRFS: Transaction aborted (error -5)
[ 3177.740034] Modules linked in: ipt_REJECT nf_reject_ipv4 br_netfilter xt_connmark iptable_mangle bnep bluetooth snd_soc_simple_card brcmfmac brcmutil cfg80211 omap_rng tilcdc rng_core rfkill snd_soc_davinci_mcasp snd_soc_edma spi_omap2_mcspi uio_pdrv_genirq uio sch_fq_codel
[ 3177.764601] CPU: 0 PID: 551 Comm: btrfs-transacti Not tainted 4.4.9+ #1
[ 3177.771242] Hardware name: Generic AM33XX (Flattened Device Tree)
[ 3177.777391] [<c001afa0>] (unwind_backtrace) from [<c0015308>] (show_stack+0x20/0x24)
[ 3177.785178] [<c0015308>] (show_stack) from [<c054bd64>] (dump_stack+0x94/0xa8)
[... cut ...]
01.09.16 09:39:41 [+0300] Downloading application 'registry.resinstaging.io/beaglebone/812e037e4caa77e48e05bd8f8e6c14e18b3f04d0'
01.09.16 09:46:06 [+0300] Failed to download application 'registry.resinstaging.io/beaglebone/812e037e4caa77e48e05bd8f8e6c14e18b3f04d0' due to 'Error pulling image (latest) from registry.resinstaging.io/beaglebone/812e037e4caa77e48e05bd8f8e6c14e18b3f04d0, failed to register layer: rename /var/lib/docker/image/btrfs/layerdb/tmp/layer-612867751 /var/lib/docker/image/btrfs/layerdb/sha256/eb6c11b92c040eb052349ce00186d8d2509662ff4fb44be1a415e7dcdd2a13ff: directory not empty'

floion, 2016-09-01:
The exact problem in the device log was:

01.09.16 09:06:59 [+0300] Failed to install application 'registry.resinstaging.io/beaglebone/812e037e4caa77e48e05bd8f8e6c14e18b3f04d0' due to 'chown /var/lib/docker/btrfs/subvolumes: read-only file system '
01.09.16 09:06:59 [+0300] Failed to update application 'registry.resinstaging.io/beaglebone/812e037e4caa77e48e05bd8f8e6c14e18b3f04d0' due to 'chown /var/lib/docker/btrfs/subvolumes: read-only file system '

The above issue with the rename is after a reboot when btrfs is rw as it should be and is tries to redownload the app

agherzan, 2016-09-01:
Looks very similar to what @shaunmulligan saw a couple of days ago. Seems like the btrfs partition gets mounted ro.

floion, 2016-09-01:
Is there an issue already opened on this? Where was he experiencing it? Staging or production? What host OS version?

The latest errors I have pasted today have been on staging, host OS 1.11.0

shaunmulligan, 2016-09-01:
@floion this is the issue we had previously: https://github.com/resin-io/hq/issues/295

PR #34 (2016-09-06): Update kernel to 4.4.19
@agherzan @telphan @michal-mazurek @lorenzo-stoakes 

fixes #25
closed 2016-09-06 by floion