balena-os/balena-beaglebone #137: Support for Beaglebone Black wireless?

FRVisser, 2018-01-16:
Hi!

Any change there will be support for the Beaglebone Black wireless? 

Thanks,
Fokko

agherzan, 2018-01-22:
Hi @FRVisser we don't have this as a priority for now. This is due to the fact that we have many other boards in progress right now. I would encourage you to try to workout patches if this is something your use-case depends on and we will try to integrate them in the next releases.

FRVisser, 2018-01-25:
Hi @agherzan I did try to patch. It seems to me (but I know nothing about the other Resin components) the only thing needed is adding the board layout to boot (which I did) and edit the uENv.txt to select it. It boot's fine and connect's to the internet over wifi and shows up, but doesn't connect, in the Dashboard. (probably because it is a Beaglebone Black application). I have it on staging right now, and If you like I can leave it running for you.

agherzan, 2018-01-25:
Thanks @FRVisser for the additional information. @floion please take a look at this when you have a chance.

floion, 2018-01-25:
@shaunmulligan shall we add this board to our list too? As in, shall we get it in-house and look at it?

shaunmulligan, 2018-01-25:
@floion yeah sounds like a good idea. I expect it to just work, but lets order some

floion, 2018-01-25:
@FRVisser can you point us to the exact hw revision you are using?

FRVisser, 2018-01-30:
Hi @floion, on the silkscreen Rev A5 is printed. I now see, the device online in the Dashboard btw and I can see now the capemgr isn't running. I'll send you the device link in a personal message.

floion, 2018-02-01:
thanks @FRVisser 
@shaunmulligan so shall we order the exact same revision?

shaunmulligan, 2018-02-01:
@floion , yeah sounds like a good idea

floion, 2018-02-07:
@FRVisser can you give an update? Last time we talked this started working. did you reflash to check again?

FRVisser, 2018-02-08:
@floion ai, it got on my todo list 😔. Give me an hour or 2, then it will be on my done list 👍

FRVisser, 2018-02-08:
So, I reflashed it and it connected to the dashboard immediately. I even didn't edit the `uENv.txt` file and only added the `am335x-boneblack-wireless.dtb` to `resin-flashA/boot/` and after it was flashed to the `/resin-rootA/balena/.../_data/` folder.

Capemgr is missing, but it could be because of the `dtb` file I used. Just pulled it of the internet form Debian I believe. Link:[am335x-boneblack-wireless.dtb](http://ftp.debian.org/debian/dists/sid/main/installer-armhf/current/images/device-tree/am335x-boneblack-wireless.dtb)

floion, 2018-02-09:
Thanks. Why did you copy it to the data folder after flashing? that should not be needed IMO

FRVisser, 2018-03-27:
The `am335x-boneblack-wireless.dtb` doesn't copy over. Only the `uEnv.txt` does.

FRVisser, 2018-04-24:
Is it possible to just add `/boot/am335x-boneblack-wireless.dtb` to the image file?

floion, 2018-04-24:
Yeah, that's what I plan to do when I have some time to look at this. Had other things in the queue and hope to get this done this week sometime.

floion, 2018-04-26:
Hi, on what OS version were you able to have the device online in the dashboard?

floion, 2018-04-26:
Also, were you connecting to a hidden network?

floion, 2018-04-26:
Should be fixed by https://github.com/resin-os/resin-beaglebone/pull/142

floion, 2018-04-26:
2.12.5+rev1 with the above PR merged, has been deployed to our staging environment. Can you try it on your side and let us know if it works for you?
Thanks

FRVisser, 2018-05-03:
@floion I currently have it on Resin OS 2.9.6+rev1 (dev). It showed up in the dashboard without problems. I didn't test a hidden network.

I'll test 2.12.5+rev1 today. Saw it passing by in the release notes.
Thanks

floion, 2018-05-24:
Hi @FRVisser did you have a chance to check this? Thanks

FRVisser, 2018-05-25:
@floion Yes it works flawlessly, including the cape-manager. 
Thanks!

floion, 2018-05-25:
Good. Closing this item

closed 2018-05-25 by floion