balena-os/balena-beaglebone #141: U-Boot Overlays / Cape Manager

bradstewart, 2018-03-21:
It seems the Kernel Overlays and Cape Manager are now deprecated in favor of U-Boot overlays:
- https://groups.google.com/d/msg/beagleboard/P_Y5yjJyuu4/G0J70kZoAwAJ

Related reading:
 - https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
 - https://groups.google.com/forum/#!topic/beagleboard/Xv9H7wxDRAE

It seems to be under active development, and require new/specific versions of U-Boot and the bb-cape-overlays package. This is way outside my area of expertise, so playing around with `uEnv_internal.txt` didn't yield much.

I suspect https://github.com/resin-os/resin-beaglebone/issues/138 is related to this as well.

Is this supported new way of doing things supported on ResinOS?

floion, 2018-03-22:
Hi @bradstewart can you set-up your cape by using the info here? https://forum.digikey.com/t/all-beaglebone-varients-u-boot-overlays/26
The beaglebones we have on v2.12.3 are running the 2017.01 u-boot version

floion, 2018-03-22:
Can you try adding "enable_uboot_overlays=1" to uEnv.txt and then
> cat /proc/cmdline

to check if it contains
> bone_capemgr.uboot_capemgr_enabled=1

?

bradstewart, 2018-03-26:
Will give it a shot today.

FRVisser, 2018-03-27:
I do have the capamgr working on v2.12.3 with UART 2, 4 and 5.

lekkas, 2018-05-17:
@bradstewart did you have any progress with enabling uboot?

bradstewart, 2018-05-30:
I tried once more, was still having trouble with it. Then life happened, we deployed a workaround instead, and I haven't had time to circle back on it yet.

floion, 2019-01-16:
Implemented by https://github.com/balena-os/balena-beaglebone/pull/171
These changes will be in OS version v2.29.2+rev1 or newer

closed 2019-01-16 by floion

jadonk, 2021-01-26:
> 
> 
> Can you try adding "enable_uboot_overlays=1" to uEnv.txt and then
> 
> > cat /proc/cmdline
> 
> to check if it contains
> 
> > bone_capemgr.uboot_capemgr_enabled=1
> 
> ?

Where do you edit uEnv.txt in a running image? I tried /mnt/boot/uEnv.txt, but the setting didn't show up.