balena-os/balena-beaglebone #144: Kernel audit seems to be enabled?

pcarranzav, 2018-06-12:
Looking at journal logs for a 2.7.5 device, there were a lot of "AUDIT" messages in the logs. It would be good to check if latest versions also have this enabled, and disable it since it only floods the logs.

ZubairLK, 2018-08-28:
yea it is.
```
root@5a524b5:~# cat /proc/config.gz | gunzip | grep -i audit
CONFIG_AUDIT=y
CONFIG_HAVE_ARCH_AUDITSYSCALL=y
CONFIG_AUDITSYSCALL=y
CONFIG_AUDIT_WATCH=y
CONFIG_AUDIT_TREE=y
CONFIG_NETFILTER_XT_TARGET_AUDIT=m
CONFIG_SECURITY_TOMOYO_MAX_AUDIT_LOG=1024
CONFIG_INTEGRITY_AUDIT=y
CONFIG_AUDIT_GENERIC=y
# CONFIG_AUDIT_ARCH_COMPAT_GENERIC is not set
root@5a524b5:~# 
```

I think it is coming from 

```
zubairlk@zubair-xps-resin:~/resin/yocto/resin-beaglebone/layers$ search audit meta-resin-beaglebone/
meta-resin-beaglebone/recipes-kernel/linux/linux-beagleboard-4.14/defconfig:224:CONFIG_NETFILTER_XT_TARGET_AUDIT=m
zubairlk@zubair-xps-resin:~/resin/yocto/resin-beaglebone/layers$ 
```

floion, 2019-04-25:
Fixed by https://github.com/balena-os/balena-beaglebone/pull/183

closed 2019-04-25 by floion