balena-os/balena-beaglebone #205: x11-window-manager demo not starting up properly (on 2.41.1+rev1)

floion, 2019-09-14:
The example demo is not starting up on the screen on version 2.41.1+rev1
https://github.com/balena-io-playground/x11-window-manager

acostach, 2020-04-02:
Same on 2020.04.0.prod esr

acostach, 2020-07-23:
Present on 2.53.8+rev1:
```
23.07.20 18:15:01 (+0300)  main  STARTING X
23.07.20 18:15:07 (+0300)  main  hostname: No address associated with hostname
23.07.20 18:15:10 (+0300)  main  xauth:  file /root/.Xauthority does not exist
23.07.20 18:15:12 (+0300)  main  xauth: (stdin):1:  bad display name "214ee2b:0" in "add" command
23.07.20 18:15:12 (+0300)  main  
23.07.20 18:15:13 (+0300)  main  
23.07.20 18:15:13 (+0300)  main  X.Org X Server 1.19.2
23.07.20 18:15:13 (+0300)  main  Release Date: 2017-03-02
23.07.20 18:15:13 (+0300)  main  X Protocol Version 11, Revision 0
23.07.20 18:15:13 (+0300)  main  Build Operating System: Linux 4.9.0-8-armmp-lpae armv7l Debian
23.07.20 18:15:13 (+0300)  main  Current Operating System: Linux 214ee2b 5.4.47+ #1 SMP PREEMPT Thu Jul 23 09:01:20 UTC 2020 armv7l
23.07.20 18:15:13 (+0300)  main  Kernel command line: console=ttyO0,115200n8 root=UUID=ba1eadef-f71d-4369-930c-6d271d62ba75 ro rootfstype=ext4 rootwait console=null quiet splash vt.global_cursor_default=0 consoleblank=0
23.07.20 18:15:13 (+0300)  main  Build Date: 03 November 2018  05:53:43AM
23.07.20 18:15:13 (+0300)  main  xorg-server 2:1.19.2-1+deb9u5 (https://www.debian.org/support) 
23.07.20 18:15:13 (+0300)  main  Current version of pixman: 0.34.0
23.07.20 18:15:13 (+0300)  main         Before reporting problems, check http://wiki.x.org
23.07.20 18:15:13 (+0300)  main         to make sure that you have the latest version.
23.07.20 18:15:13 (+0300)  main  Markers: (--) probed, (**) from config file, (==) default setting,
23.07.20 18:15:13 (+0300)  main         (++) from command line, (!!) notice, (II) informational,
23.07.20 18:15:13 (+0300)  main         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
23.07.20 18:15:13 (+0300)  main  (==) Log file: "/var/log/Xorg.0.log", Time: Thu Jul 23 15:15:13 2020
23.07.20 18:15:13 (+0300)  main  (==) Using system config directory "/usr/share/X11/xorg.conf.d"
23.07.20 18:15:14 (+0300)  main  gbm: failed to open any driver (search paths /usr/lib/arm-linux-gnueabihf/dri:${ORIGIN}/dri:/usr/lib/dri)
23.07.20 18:15:14 (+0300)  main  gbm: Last dlopen error: /usr/lib/dri/tilcdc_dri.so: cannot open shared object file: No such file or directory
23.07.20 18:15:14 (+0300)  main  failed to load driver: tilcdc
23.07.20 18:15:17 (+0300)  main  EGL_MESA_drm_image required.
```

acostach, 2020-08-11:
Present on v2020.07.1