balena-os/balena-beaglebone #229: Beaglebone Green instructions don't tell the user to hold the USER button down when provisioning

shaunmulligan, 2020-06-25:
A customer was struggling to provision a BBG, it seems we changed the instructions incorrectly and they no longer tell the user to hold down the `USER` button to force the device to boot from SD card initially. It seems it was changed here: https://github.com/balena-os/balena-beaglebone/commit/0e4e7819dbc119b5118a58d44d04118b69bfab16

acostach, 2020-06-25:
Interesting @shaunmulligan , there shouldn't be a need to hold down the boot button on the BBG, it only needs to be pressed if an SD-card is inserted, but it's desired to boot from eMMC. This is how the BBGs we have behave, and [docs](https://wiki.seeedstudio.com/BeagleBone_Green/) say: 

```
Boot button

    When there's a SD card insert, the system will boot from SD card first, if you want to boot from eMMC, press this button and then power on.
```

I wonder if this could be a case of faulty hardware, as beagleboard.org confirms: https://beagleboard.org/getting-started

```
Insert SD card into your (powered-down) board, hold down the USER/BOOT button (if using Black) and apply power, either by the USB cable or 5V adapter.
```

shaunmulligan, 2020-06-26:
@acostach are you sure about that, on every BBB, BBG and BBGW I have ever provisioned it has always needed to hold down the `USER` button to force SD card boot, maybe the changed the hardware on a recent revision or updated uboot or something, but we should probably flash the default debian onto a BBG and then try flash balenaOS over it and see what happens.

shaunmulligan, 2020-06-26:
Step 8 of their wiki says the opposite of their docs https://seeeddoc.github.io/BeagleBone_Green/

acostach, 2020-06-26:
Thanks @shaunmulligan , you are correct and the docs are wrong at the point I saw. I verified with a debian image on the emmc and indeed it's necessary to hold it down. If Balena is already flashed, the issue is masked. I've updated my PR to reflect this and will get it to the dashboard.

shaunmulligan, 2020-06-26:
Thanks @acostach ! approved the change 👍

acostach, 2020-06-30:
This has been addressed in production release 2.51.1+rev2.

closed 2020-06-30 by acostach