balena-os/balena-beaglebone #580: Adding beagleplay support: mkfs.ext4: Could not allocate block in ext2 filesystem while populating file system

delhiryder, 2023-08-15:
This error is happening every time I try and create a `balena-image` target for the newly added `beaglebone-play` machine (which is actually an alias for the `beagleplay` machine (which ti added to their `meta-ti-bsp` repo earlier this summer).

The failing recipe:task in question is `balena_image:do_image_hostapp_ext4`:

```
...
time="2023-08-15T22:36:58.658711617Z" level=info msg="Loading containers: done."
time="2023-08-15T22:36:58.712299067Z" level=warning msg="failed to retrieve balena-engine-init version: exec: \"balena-engine-init\": executable file not found in $PATH"
time="2023-08-15T22:36:58.719222824Z" level=info msg="Docker daemon" commit=8f374c29395adff1f6f7796b06418e593a479a7d graphdriver(s)=overlay2 version=v20.10.38
time="2023-08-15T22:36:58.719252405Z" level=info msg="Daemon has completed initialization"
time="2023-08-15T22:36:58.751827226Z" level=info msg="API listen on /var/run/balena-host.sock"
+ hostapp-update -f /input -n
/mnt/sysroot/inactive/balena/tmp is not a mountpoint
+ kill 7
+ wait 7
time="2023-08-15T22:37:11.800296016Z" level=info msg="Processing signal 'terminated'"
time="2023-08-15T22:37:11.801934131Z" level=info msg="stopping event stream following graceful shutdown" error="<nil>" module=libcontainerd namespace=moby
time="2023-08-15T22:37:11.802399994Z" level=info msg="Daemon shutdown complete"
time="2023-08-15T22:37:11.802430342Z" level=info msg="stopping healthcheck following graceful shutdown" module=libcontainerd
time="2023-08-15T22:37:11.802482483Z" level=info msg="stopping event stream following graceful shutdown" error="context canceled" module=libcontainerd namespace=plugins.moby
+ mkfs.ext4 -F -E lazy_itable_init=0,lazy_journal_init=0 -i 8192 -d /mnt/sysroot/inactive /output
mke2fs 1.46.5 (30-Dec-2021)
Discarding device blocks: done
Creating filesystem with 348160 1k blocks and 43688 inodes
Filesystem UUID: 056f2d08-c5e1-430e-b581-ee66b8acb663
Superblock backups stored on blocks:
        8193, 24577, 40961, 57345, 73729, 204801, 221185

Allocating group tables: done
Writing inode tables: done
Creating journal (8192 blocks): done
Copying files into the device: __populate_fs: Could not allocate block in ext2 filesystem while writing file "Image"
mkfs.ext4: Could not allocate block in ext2 filesystem while populating file system
Untagged: mkfs-hostapp-native:1691690000
Deleted: sha256:d0e5718d715f3f91ec0f201d459e2f978de3aa077e38256094f55d18198e1b7f
WARNING: exit code 1 from a shell command.
```

I have added the warning line in the script (at timestamp: `time="2023-08-15T22:36:58.712299067Z"`) to ensure this is not the cause of the failure (looks like red herring to me).

The **actual** failure appears to be the one caused by the `mkfs.ext4` script.

Any thoughts, ideas ?

Regards,

Sidd

delhiryder, 2023-08-16:
FYI, I managed to get aroud this problem by **hard-coding** the `IMAGE_ROOTFS_SIZE` values to `655360` (this is probably not the right way to get around this issue).

I added the following lines to the `layers/meta-balena-beaglebone/recipes-core/images/balena-image.bbappend` file:

```
IMAGE_ROOTFS_SIZE:beaglebone-play = "655360"
IMAGE_ROOTFS_SIZE:beaglebone-play-k3r5 = "655360"
```

The image now builds properly (It's size is almost 1.6 GB, but that is a problem for another day...)

closed 2023-08-16 by delhiryder

floion, 2023-08-17:
Hi, actually that is the correct way to handle a bigger rootfs size. We do it for other boards, for example https://github.com/balena-os/balena-radxa/blob/master/layers/meta-balena-rockpi/recipes-core/images/balena-image.inc#L9-L13

You may want to lower that size though. What is the generated rootfs size? Typically, I add around 20-30% extra, not more.